Northeastern-Electric-Racing/Embedded-Base #54: [DevOps] - Streamline Windows Docker Start

nwdepatie, 2024-01-05:
### Description

Right now working with the docker container is very much doable for developers, but for electrical engineers who don't know about our tooling it can be a bit frustrating. We should create a script that can be run on windows to auto-launch a lot of this stuff, in addition to mounting the RPi debug probe

### Acceptance Criteria

Script that runs on windows that will start the docker container and give users a CLI out of the box

### Proposed Solution

Probably via a batch file, maybe via Python since we already are using python for auto mounting the RPi Debug Probe to the container

nwdepatie, 2024-01-05:
Stretch:  Install Python3 and Docker for user if they don't have them installed? (probably using winget)

dyldonahue, 2024-07-19:
closing as likely going to be solved by imporvements to dev environemnt

closed 2024-07-19 by dyldonahue