Northeastern-Electric-Racing/Embedded-Base #105: [DevOps] - Fix Git in Docker

dyldonahue, 2024-03-28:
### Current Features

Currently git is kinda broken in the docker container, as it doesn't use the credentials that are on the host machine. So, a user needs to exit the docker container to run git commands.

### Desired Additional Features

We need to implement a solution that will carry over git credentials and ssh keys from the host machine into the docker container, that way developers can remain in the docker container for the entirety of development rather than having to exit to commit changes

jr1221, 2024-03-28:
Ill try and take care of this I have already research this for my docker container.

dyldonahue, 2024-07-19:
@jr1221 closing as likely impossible & possibly redundant if people like the python venv setup :)

closed 2024-07-19 by dyldonahue