Northeastern-Electric-Racing/Embedded-Base #173: Develop stable low pass filter algorithms for voltages and temperatures

dyldonahue, 2024-10-10:
### Current Features

We have used various instable, volatile, and aggressive algorithms to eliminate noise form our data readings. These are case-specific to behavior weve seen, not strong agaisnt edge cases, and generally over aggresive

### Desired Additional Features

We should create stable low pass filtering to eliminate high frequency noise. This should be a simple and generalized filter that can be applied to data of any kind. It should not be especially aggressive and should just work as a no brainer to apply to any incoming data

jr1221, 2024-10-10:
Check out cerberus/dti.c for examples of low pass filters.

jr1221, 2024-10-10:
https://github.com/Northeastern-Electric-Racing/Cerberus/blob/39d056a668c43d7f23e70a3f341a94cb17d78e98/Core/Src/dti.c#L46

PR #210 (2025-01-14): Feature/lowpass filter
## Changes

Added the simple_ema.h file and renamed branch

## Notes

Alpha works well for 0.1

## Test Cases

- Case A
- Edge case
- ...

## To Do

_Any remaining things that need to get done_

- [ ] item 1
- [ ] ...

## Checklist

It can be helpful to check the `Checks` and `Files changed` tabs.
Please reach out to your Project Lead if anything is unclear.
Please request reviewers and ping on slack only after you've gone through this whole checklist.

- [ *] All commits are tagged with the ticket number
- [* ] No merge conflicts
- [ *] All checks passing
- [* ] Remove  [... cut]
closed 2025-01-14 by DimitryP6